microsoft/graphics-driver-samples #13: How to use this ?

matangpanchal, 2015-12-05:
Hey Senior Programmers,
I am not sure , if this is the right place to discuss this or not, please forgive if I am wrong somewhere.

I build this drivers and installed on raspberry pi 2 as per the instructions given on wiki page. 
Then I deployed my app with some storyboard animation like scale, translate , fade animations , but not able to see any difference yet.
Should I see any difference in my above scenerio ? or this Drivers are ment for something else ?
If this is for my purpose , how can I take advantage of this ?

Please guide me.

Regards,
Matang

bhouse-microsoft, 2015-12-07:
Hi Matang,

It's totally awesome that you were able to build and install the driver on the Pi 1.  Unfortunately, the driver is not ready for more general use yet.  We are just getting a very basic shader compiler up and running at the moment so the driver is not really ready for general DX API usage.  We will update the README.

Bart

bhouse-microsoft, 2016-01-06:
In theory and depending upon the application you were trying and the content being animated, this release should help improve use of XAML controls.

If you give us a pull request with the source of your story board app placed at render-only-sample\tests\story-board-app then we can test against your application when we are working on the R3 release.

matangpanchal, 2016-01-08:
Dear Sir,
Thanks you for considering my question and doing follow up as well, As per your instruction I have made a pull request here. 
Kindly do the needfull.

fbsanches, 2017-11-09:
Has the project died? No chance iot rotate videos on raspberry pi 3?