microsoft/graphics-driver-samples #25: Make sure texture formats needed for Direct2D/XAML rendering are support

indygit, 2016-01-26:
At least, we need:
DXGI_FORMAT_R8G8B8A8_UNORM
DXGI_FORMAT_R8G8_UNORM,
DXGI_FORMAT_R8_UNORM
DXGI_FORMAT_A8_UNORM

kika123, 2016-01-27:
HVS_PIXEL_FORMAT_RGB332 = 0,	
HVS_PIXEL_FORMAT_RGBA4444 = 1,
HVS_PIXEL_FORMAT_RGB555 = 2,
HVS_PIXEL_FORMAT_RGBA5551 = 3,
HVS_PIXEL_FORMAT_RGB565 = 4,
HVS_PIXEL_FORMAT_RGB888 = 5,
HVS_PIXEL_FORMAT_RGBA6666 = 6,
HVS_PIXEL_FORMAT_RGBA8888 = 7
#define HVS_PIXEL_ORDER_RGBA			0
#define HVS_PIXEL_ORDER_BGRA			1
#define HVS_PIXEL_ORDER_ARGB			2
#define HVS_PIXEL_ORDER_ABGR			3
#define HVS_PIXEL_ORDER_XBRG			0
#define HVS_PIXEL_ORDER_XRBG			1
#define HVS_PIXEL_ORDER_XRGB			2
#define HVS_PIXEL_ORDER_XBGR			3
#define HVS_PIXEL_ORDER_XYCBCR			0
#define HVS_PIXEL_ORDER_XYCRCB			1
#define HVS_PIXEL_ORDER_YXCBCR			2
#define HVS_PIXEL_ORDER_YXCRCB			3

kika123, 2016-01-27:
That are the formats that the hardware scaler and layers understand

hideyukn88, 2016-01-27:
Basically 3 kind of formats support in graphics driver, one for texture sampling (where texture look up unit can read at pixel shader), and render target format (where pixel shader output to tile buffer and then stored to final RT memory), then one that HVS (hardware video scaler) can handle for display output. That said, not all HVS supported format can be available at texture look up unit (vise-versa). This work item is focusing on supporting various formats for texture sampling for satisfying (some basic) XAML application to run. If VC4 hardware doesn't meet that (minimum) requirement, it's possible that this work includes some sort of emulation by converting to supported format underneath.

kika123, 2016-01-27:
You can abuse the HVS to render to memory then use it as a texture

indygit, 2016-01-29:
Hi, Marek:

    I assigned #25, since we already discussed it for #22.

                                 Thanks, Indy, 1/29/2016

indygit, 2016-01-29:
Hi, Marek:

    I assigned this one to you, since we already discussed it for issue #22.

                                Thanks, Indy, 1/29/2016

marekkedzierski, 2016-02-14:
All, 
Please take a look at [](url) https://github.com/marekkedzierski/graphics-driver-samples/commit/397d7c2a31d48031757127d46bf465394560ed0e
This is initial support for 32/16/8bpp tiled textures. This is not finished, but please take a look and do a short review. Some values are hardcoded only to see if those types of color depths are rendererd properly.

I have a question regarding formats:
DXGI_FORMAT_R8G8_UNORM,
DXGI_FORMAT_R8_UNORM
DXGI_FORMAT_A8_UNORM

Is their layout and format the same as CL_RG, CL_R and CL_A from https://www.khronos.org/registry/cl/sdk/1.0/docs/man/xhtml/cl_image_format.html ? I need a confirmation.

I've modified CubeTest a bit (no commit yet) to convert initial texture to different formats and render them. So I need information how to handle given formats internaly (conversion and handling).

Thanks

Marek

hideyukn88, 2016-02-16:
First, for D3D11 and OpenGL texture format mapping, I would suggest that you reference ANGLE project at https://github.com/MSOpenTech/angle, particularly angle\src\libANGLE\Renderer\d3d\d3d11\texture_format_table.cpp. Second, regarding to Khronos spec, what you pointed is for OpenCL, not OpenGL, but yes, those are basically same format as long as unnormalized unsigned 8-bit integer value per channel (CL_UNSIGNED_INT8).

indygit, 2016-02-16:
Thanks Marek. I will take a look at your initial change and report back.

Cheers, Indy

indygit, 2016-02-19:
Hi, Marek:

   Your initial change marekkedzierski@ 397d7c2 looks good to me.

   If you haven't checked in your tests (for issue #22, #25) yet, please check them in. I think it is better to keep tests that have been proven working around.

                                   Thanks, Indy, 2/18/2016