microsoft/graphics-driver-samples #35: Implement Raspberry Pi support for RelocateDeviceFuncs(D3D11_1)

jeffwick-msft, 2016-02-10:
Support for RelocateDeviceFuncs(D3D11_1) needs to be implemented in the Raspberry Pi driver in order to support Windows Shell (SIHost). This is currently stubbed at:

rosumd!RosUmdDeviceDdi::RelocateDeviceFuncs11_1_Default

MSDN info: https://msdn.microsoft.com/en-us/library/windows/hardware/hh439830(v=vs.85).aspx

jordanrh1, 2016-03-09:
Looking at the WARP implementation, this function can probably be a no-op.

hideyukn88, 2016-03-09:
Driver has to update a pointer to function table (one of blow) with new pointer given from this DDI. Some driver does not cache this pointer, so they don't need to update. While ROSUMD doesn't use this pointer (to patch/redirect DDI call from runtime), but it still does cache it in local class (at below), so it's better to update the pointer (to keep in sync with runtime, this would help debugging, too)

class RosUmdDevice
...
union
{
    D3D10DDI_DEVICEFUNCS*       m_pDeviceFuncs;
    D3D10_1DDI_DEVICEFUNCS*     m_p10_1DeviceFuncs;
    D3D11DDI_DEVICEFUNCS*       m_p11DeviceFuncs;
    D3D11_1DDI_DEVICEFUNCS*     m_p11_1DeviceFuncs;
    D3DWDDM1_3DDI_DEVICEFUNCS*  m_pWDDM1_3DeviceFuncs;
    D3DWDDM2_0DDI_DEVICEFUNCS*  m_pWDDM2_0DeviceFuncs;
};