microsoft/graphics-driver-samples #48: Access violation in rosumd!RosUmdResource::Form1kSubTileBlock

jordanrh1, 2016-03-22:
An access violation is seen in sihost.exe when IotShellExtension attemps to load the splash screen bitmap.

Repro steps:
 1. Copy over the stock `IotShellExt.dll`, not the one in render-only-sample/scripts
 2. Enable initial breakpoint in windbg (ctrl+alt+k)
 3. Boot machine. When initial breakpoint is hit, run the following windbg commands to enable break on usermode exceptions

        sxe eh
        sxe av
        !gflag +sue
        !gflag +soe

Resume booting (hit 'g'). The following access violation will occur

```
Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
VCRUNTIME140D!__memcpy_forward_large_neon+0x30:
752b1430 680c     ldr         r4,[r1]

1: kd> r
r0=01b80000  r1=00d55ae0  r2=00000008  r3=01b80000  r4=75427345  r5=00307010
r6=0008e800  r7=0008dd50  r8=002abc7c  r9=00307100 r10=002ac838 r11=0008dc60
r12=752c5004  sp=0008dc58  lr=754297c1  pc=752b1430 psr=20000030 --C-- Thumb
VCRUNTIME140D!__memcpy_forward_large_neon+0x30:
752b1430 680c     ldr         r4,[r1]                         00d55ae0=????????

1: kd> k
00 0008dca0 754298d4 rosumd!RosUmdResource::Form1kSubTileBlock [d:\workspace\graphics-driver-samples\render-only-sample\rosumd\rosumdresource.cpp @ 484]
01 0008dca0 75429748 rosumd!RosUmdResource::Form4kTileBlock+0xf0 [d:\workspace\graphics-driver-samples\render-only-sample\rosumd\rosumdresource.cpp @ 563]
02 0008dce0 75423c64 rosumd!RosUmdResource::ConvertBitmapTo4kTileBlocks+0xd4 [d:\workspace\graphics-driver-samples\render-only-sample\rosumd\rosumdresource.cpp @ 596]
03 0008dd28 75427372 rosumd!RosUmdDevice::CreateResource+0x3b0 [d:\workspace\graphics-driver-samples\render-only-sample\rosumd\rosumddevice.cpp @ 336]
04 0008de90 76016b3a rosumd!RosUmdDeviceDdi::DdiCreateResource+0x2e [d:\workspace\graphics-driver-samples\render-only-sample\rosumd\rosumddeviceddi.cpp @ 249]
[private symbols redacted]
```

jordanrh1, 2016-03-22:
Hi Marek, I'd appreciate if you could investigate this issue as it seems related to tiled texture support. Thanks,
Jordan

marekkedzierski, 2016-03-23:
OK, I'll take a look at this.

hideyukn88, 2016-03-27:
Just FYI: I disabled tiled texture until this get fixed. See above commit.

marekkedzierski, 2016-03-27:
How to reproduce this issue? What is the easiest way to run iotshellext? Where can I get IotShellExt.dll (if it is not from render-only-sample/scripts). Thanks, Marek.